NETMF/llilum #187: LLVM Messages during initial compile

piwi1263, 2016-02-22:
After getting the LLVM 3.8.0 RC1 sources and (hopefully) followed the instructions from the setup in the wiki and using the BuildLlvmWithVs.cmd some unexpected messages displayed, did I miss a setup step ? Did anybody ran into this before ?

[build.zip](https://github.com/NETMF/llilum/files/141633/build.zip)

cw2, 2016-02-23:
What do you mean by 'unexpected messages'? There is nothing unusual in the build.log, those are standard output messages, safe to ignore them...

piwi1263, 2016-02-23:
I wouldn't expect the warning about not finding file Compiler/MSVC-ASM as other files not found. 

But LLVM build successfully, as well as the SDK, MSI, VSIX and the install works.

Thought maybe this could be the cause of the Native project not to build when selecting  STM32F411 in the default project template, since I got the following message:

[image]

piwi1263, 2016-02-23:
As another for me hard to explain list of 1575 errors ??

[image]