NETMF/llilum #190: Too many breakpoints

miloush, 2016-02-24:
Setting too many breakpoints results in an error box saying that hardware breakpoints could not be inserted (>6 for LPC1768). At minimum this should result in the breakpoints that could be set to work, preferably those that could not be set to be disabled or 'not resolved', and ideally this should be handled and not happen.

smaillet-ms, 2016-02-26:
I don't know what #66 is (the description is blank) but the physical chip has a limited set of breakpoints and unless the code is running in RAM we can't replace instruction with a software breakpoint. So, Once the limit is hit, we have to let you know that it can't set any more.

miloush, 2016-02-26:
I don't know either, I guessed potential relevance from the title.
Good point about not running from RAM. In that case I think the best option would be to leave the breakpoints that could have not been set as unresolved. The dialog box is not extremely useful when you set the breakpoints first and then try to run the program, as you don't know which breakpoints will be hit and which not.