NETMF/llilum #191: Breakpoints not hit (and deleted ones hit)

miloush, 2016-02-24:
I am running the default SDK application on LPC1768 and I cannot hit any breakpoints until inside the while loop (both while loops work). 

- Starting the application with F10 results in running like F5 was pressed
- Some breakpoints are never hit even inside the while loop, eg. on the if statement inside the loop.
- Some breakpoints are hit even after deleting them during debugging.

I did have F10 previously working with the same board on another computer, that was probably an older version though (and had #172).

lt72, 2016-02-26:
Steve, can you take a look when you have a chance?

smaillet-ms, 2016-02-26:
Are you using the -Debug switch for the compiler in the frontend .config file? To get the best full debugging experience you will need that. The symptoms you describe sound like common ones due to the LLVM optimizations and inherent limits of the DWARF-2 format for the debugger build with the ARM Embedded tools.

miloush, 2016-02-26:
As I said this is SDK application, so I have not modified the .config files (kind of expecting the Debug/Release solution configuration would have a say). Anyway, I tried to add -Debug option after -Architecture in mbed_simple_LPC1768.FrontEndConfig which is part of the Native project, rebuilt, and it did not seem to have any effect. The build output however does not say if it used that file or not...

smaillet-ms, 2016-02-26:
At this point the configuration settings for the SDK are reflected only to the Roslyn compiler or the C/C++ compiler. We have a very simplistic MSBuild Exec task to run the Llilum specific tools at the moment. However the goal is to replace that with property tasks that can expose all the command line functionality and information currently stored in the frontend .config files. (Ideally those will fade out of use over time in favor of real project settings)