NETMF/llilum #195: wiki issues

cyberh0me, 2016-02-26:
related to [https://github.com/NETMF/llilum/wiki/demo](url)
1) -cfg ..\..\..\..\Zelig\CompileTime\CodeGenerator\FrontEnd\mbed_simple.FrontEndConfig
its the wrong path from > <repo>\Zelig\LLVM2IR_results\mbed\simple\ (one ..\ to much)
2) <repo>\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd does not contain a file named  mbed_simple.FrontEndConfig
3) it seems there is an issue with Microsoft.Zelig.Test.mbed.Simple_opt.o, build needs it but it does not exist

lt72, 2016-02-26:
re: 3)
the file is generated after building with the Zelig compiler (i.e. launching FrontEnd and letting it finish throw LLVM codegen)
the part of the FrontEnd configuration that that causes the FrontEnd tool to emit the object file is 
`-GenerateObj` in the [FronEnd config file](https://github.com/NETMF/llilum/blob/dev/Zelig/Zelig/CompileTime/CodeGenerator/FrontEnd/mbed_simple_LPC1768.FrontEndConfig)

cyberh0me, 2016-02-26:
and which file should i use instead of the missing one from point 2?

lt72, 2016-02-27:
Hi cyberh0me,

if you are working with an LPC1768 board, you should use 
`E:\src\iot\forks\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\mbed_simple_LPC1768.FrontEndConfig`

is that your board of choice?

if not, use
`E:\src\iot\forks\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\mbed_simple_XXX.FrontEndConfig`

where XXX is your board... 

there are a few other files where XXX is NET_XXX or XXX_Tester. Those are for using the TCP/IP stack or running the test harness

lt72, 2016-02-27:
please re-open as necessary...

cyberh0me, 2016-03-01:
i try to use the nucleo stm32l152re
anyway if i try to launch the frontend i see this :-(
Caught exception: Microsoft.Zelig.TypeConsistencyErrorException: 'AbstractReferenceTypeRepresentation(Microsoft.Zelig.Runtime.Peripherals)' is marked as ForcedInstantiation, but no candidate was found
   at Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildHierarchyTables() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:line 992
   at Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildCrossReference() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:line 889
   at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Phases.PrepareImplementationOfInternalMethods.Run() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Phases\PrepareImplementationOfInternalMethods.cs:line 38
   at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.PhaseDriver.Execute() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\PhaseDrivers\PhaseDriver.cs:line 124
   at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Controller.ExecuteSteps(Boolean fGenerateImageOnly) in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Controller.cs:line 293
   at Microsoft.Zelig.FrontEnd.Bench.Compile() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 1866
   at Microsoft.Zelig.FrontEnd.Bench.RunBench(String[] args) in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 2323
   at Microsoft.Zelig.FrontEnd.Bench.<>c__DisplayClass128_0.<Main>b__0() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 2339

cyberh0me, 2016-03-01:
"please re-open as necessary..."
"you cannot re-open your own issues if a repo collaborator closed them"

lt72, 2016-03-02:
I did not know that about issues!
I am fixing the issue right now. and about to push to upstream. 
After I do, you will have to re-generate the SDK following guidelines here: https://github.com/NETMF/llilum/wiki/SDK-User-Guide

(I believe you know that already, mentioning it just in case...)

The issue is that our compiler will want to make sure that _abstract_ types marked with the 
            [SingletonFactory()]
attribute are instantiated (made concrete) in some of the sources to be compiled.
That attribute is applied to items like GarbageColletor or Processor, etc.... You must have one of those!

The Peripherals abstract type instead was never made concrete by any sub-class.

commit 3cd4e3a1 (2016-03-02): Fixes #195
closed 2016-03-02 by lt72