NETMF/llilum #202: Frontend fails to launch

cyberh0me, 2016-03-01:
Caught exception: Microsoft.Zelig.TypeConsistencyErrorException: 'AbstractReferenceTypeRepresentation(Microsoft.Zelig.Runtime.Peripherals)' is marked as ForcedInstantiation, but no candidate was found
 at Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildHierarchyTables() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:line 992
 at Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildCrossReference() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:line 889
 at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Phases.PrepareImplementationOfInternalMethods.Run() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Phases\PrepareImplementationOfInternalMethods.cs:line 38
 at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.PhaseDriver.Execute() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\PhaseDrivers\PhaseDriver.cs:line 124
 at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Controller.ExecuteSteps(Boolean fGenerateImageOnly) in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Controller.cs:line 293
 at Microsoft.Zelig.FrontEnd.Bench.Compile() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 1866
 at Microsoft.Zelig.FrontEnd.Bench.RunBench(String[] args) in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 2323
 at Microsoft.Zelig.FrontEnd.Bench.<>c__DisplayClass128_0.
b__0() in D:\Llilum\llilum.src\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 2339

lt72, 2016-03-02:
I assume this is the issue with the STM32L152 platform. Fixed with 3cd4e3a141213b98dad3894721b925e9b9fc7e53

cyberh0me, 2016-03-02:
confirmed

closed 2016-03-02 by cyberh0me