NETMF/llilum #203: simpledemo fails on STM32L152RET6

cyberh0me, 2016-03-02:
i am trying to use the STM32L152 BSP on STM32 Nucleo for L1 series board (STM32L152RET6 64 Pins) with no success, the resulting binary does not deliver the expecting output
should it work out of the box?

[Program.zip](https://github.com/NETMF/llilum/files/155214/Program.zip)

smaillet-ms, 2016-03-02:
No, the "B" in BSP is for Board and therefore is specific to a particular board. The current BSPs are unfortunately named (mostly based on the SOC in them). This is something I plan to change once I complete the shift to building and using BSP components as NUGET packages. Until then you will need to follow the [steps in the wiki] (https://github.com/NETMF/llilum/wiki/Creating-a-Board-Support-Package)
to clone and modify the current BSP to match the board you are using.

cyberh0me, 2016-03-02:
i understand, thank you

closed 2016-03-02 by cyberh0me