NETMF/llilum #213: Visual C++ for Linux Development extension (?)

cw2, 2016-04-06:
Following the announcements regarding Visual Studio and *nix development made during the Build conference and release of [Visual C++ for Linux Development extension](https://blogs.msdn.microsoft.com/vcblog/2016/03/30/visual-c-for-linux-development/), I am wondering whether that is something LLILUM and/or NETMF could benefit from?

Unfortunately, I have not yet had a chance to dig deeper into possible use cases, but it seems to touch areas mentioned in several issues (LLILUM debugging, NETMF build system, custom toolchain support and such). Is there any cooperation or plan for? Or are there any ground-breaking features to surprise us in the near future (that would explain weeks of silence)?

piwi1263, 2016-04-07:
I sincerely hope for a nice surprise !!!

Until than, business as usual ...

lt72, 2016-04-14:
Nothing planned so far, but this is definitely on our radar. Integration should be smooth enough, since it is a clean overlap and evolution from our current approach.