NETMF/llilum #223: Incorrect build&setup guide at wiki

Snarkorel, 2016-08-17:
Setup and build instructions from [wiki](https://github.com/NETMF/llilum/wiki/Setup) says, that I need to install LLVM-3.8.1 binaries and download CFE sources of LLVM 3.8.1.
But BuildLlvmWithVS.cmd script file is configured to the 3.8.0 version of LLVM, same issue for [LlvmApplicaton.props](https://github.com/NETMF/llilum/blob/dev/Zelig/Zelig/CompileTime/Llvm.NET/LibLLVM/LlvmApplication.props).
More than that, I can't build CFE sources due to error:

```
llvm-config not found -- LLVM_CONFIG-NOTFOUND

-- Configuring incomplete, errors occurred!
```

There is llvm-config in LLVM sources, not Clang. So, should I merge Clang and LLVM sources into one directory and try to build, or there is mistake in build instructions at wiki?

Snarkorel, 2016-08-17:
[Pull request](https://github.com/NETMF/llilum/pull/224) created. Please update the wiki.

Snarkorel, 2016-08-18:
[Wiki updated](https://github.com/Snarkorel/llilum-wiki), since github doesn't support pull requests for wikis, please add proposed changes.

lt72, 2016-08-20:
Fixed with 80a87a2b75f4ae248d573bd7e8f9a1cde4c453e1
thank you!

closed 2016-08-20 by lt72