NETMF/llilum #225: Issue  same as 181

nbsn2, 2016-08-25:
branch: dev

Issue: same as 181
https://github.com/NETMF/llilum/issues/181

Cause:
https://github.com/NETMF/llilum/commit/d2669ec0250a957e7dfc1240eeadfebc03bb05ef
lt72 committed on Mar 5 

Zelig/BoardConfigurations/STM32F411/Board/HardwareModel/Processor.cs 
   -    [RT.ProductFilter("Microsoft.Llilum.BoardConfigurations.STM32F411MBEDHosted")]  
 13   +    [RT.ProductFilter("Microsoft.Llilum.BoardConfigurations.STM32F411MBED")]  

 Zelig/BoardConfigurations/STM32F411/Configuration/Configuration.cs 
-    [DisplayName("STM32F411 MBED Hosted")]  
81    -    public sealed class STM32F411MBEDHosted : ProductCategory  
 80   +    [DisplayName("STM32F411 MBED")]  
 81   +    public sealed class STM32F411 : ProductCategory  

Conflict:
llilum / VisualStudio / LlilumProjectType / LlilumApplication / LlilumApplication.ProjectTemplate / LlilumNative / mbed_simple_STM32F411.FrontEndConfig 

13 #~#-CompilationSetupPath ..\..\..\..\ZeligBuild\Host\bin\Debug\Microsoft.Llilum.BoardConfigurations.STM32F411.dll 
14 -CompilationSetup Microsoft.Llilum.BoardConfigurations.STM32F411MBED**Hosted**CompilationSetup 

Hosted is still there.

nbsn2, 2016-08-26:
After manually deleted  "Hosted" in  FrontEndConfig , compling process can continue, but still no luck.
This time stuck to last phase:

`1>
1>  Compiling...
1>
1>      16.763 [   0.193]: Phase: ReduceNumberOfTemporaries                          [types:  4043,  fields:   4640,  methods:  26228]
1>      31.167 [  14.404]: Phase: TransformFinallyBlocksIntoTryBlocks                [types:  4043,  fields:   4640,  methods:  26228]
1>      34.902 [   3.735]: Phase: ApplyClassExtensions                               [types:  4043,  fields:   4640,  methods:  26228]
1>     125.165 [  90.263]: Phase: PrepareImplementationOfInternalMethods             [types:  3962,  fields:   4626,  methods:  25775]
1>     141.534 [  16.369]: Phase: CrossReferenceTypeSystem                           [types:  3962,  fields:   4720,  methods:  26024]
1>     142.541 [   1.007]: Phase: ApplyConfigurationSettings                         [types:  3962,  fields:   4720,  methods:  26024]
1>  Garbage collector selected: Microsoft.Llilum.STM32F411.GarbageCollectionManager
1>  Skipping phase: ResourceManagerOptimizations
1>     151.981 [   9.439]: Phase: HighLevelTransformations                           [types:  3962,  fields:   4720,  methods:  26024]
1>     183.457 [  31.477]: Phase: PropagateCompilationConstraints                    [types:  3962,  fields:   5127,  methods:  26024]
1>     184.038 [   0.581]: Phase: ComputeCallsClosure                                [types:  3962,  fields:   5127,  methods:  26024]
1>     198.468 [  14.430]: Phase: EstimateTypeSystemReduction                        [types:  3962,  fields:   5127,  methods:  26024]
1>     199.717 [   1.249]: Phase: CompleteImplementationOfInternalMethods            [types:  3962,  fields:   5127,  methods:  26024]
1>     200.228 [   0.511]: Phase: HighLevelTransformations                           [types:  3962,  fields:   5127,  methods:  26024]
1>     213.518 [  13.289]: Phase: PropagateCompilationConstraints                    [types:  3962,  fields:   5127,  methods:  26024]
1>     214.146 [   0.628]: Phase: ComputeCallsClosure                                [types:  3962,  fields:   5127,  methods:  26024]
1>     219.253 [   5.108]: Phase: EstimateTypeSystemReduction                        [types:  3962,  fields:   5127,  methods:  26024]
1>     219.955 [   0.702]: Phase: CompleteImplementationOfInternalMethods            [types:  3962,  fields:   5127,  methods:  26024]
1>     219.961 [   0.005]: Phase: ReduceTypeSystem                                   [types:  3962,  fields:   5127,  methods:  26024]
1>  Skipping phase: PrepareExternalMethods
1>     221.744 [   1.783]: Phase: DetectNonImplementedInternalCalls                  [types:   365,  fields:    656,  methods:   1540]
1>     221.746 [   0.002]: Phase: OrderStaticConstructors                            [types:   365,  fields:    656,  methods:   1540]
1>     221.984 [   0.238]: Phase: LayoutTypes                                        [types:   365,  fields:    656,  methods:   1540]
1>     227.781 [   5.797]: Phase: HighLevelToMidLevelConversion                      [types:   365,  fields:    656,  methods:   1540]
1>     239.798 [  12.017]: Phase: FromImplicitToExplicitExceptions                   [types:   365,  fields:    656,  methods:   1540]
1>     243.637 [   3.839]: Phase: ReferenceCountingGarbageCollection                 [types:   365,  fields:    656,  methods:   1540]
1>  Skipping phase: MidLevelToLowLevelConversion
1>  Skipping phase: ConvertUnsupportedOperatorsToMethodCalls
1>  Skipping phase: ExpandAggregateTypes
1>  Skipping phase: SplitComplexOperators
1>  Skipping phase: FuseOperators
1>     243.646 [   0.009]: Phase: Optimizations                                      [types:   365,  fields:    656,  methods:   1540]
1>  Skipping phase: ConvertToSSA
1>  Skipping phase: PrepareForRegisterAllocation
1>  Skipping phase: CollectRegisterAllocationConstraints
1>  Skipping phase: AllocateRegisters
1>     244.508 [   0.861]: Phase: GenerateImage                                      [types:   365,  fields:    656,  methods:   1540]
1>       0.000393: START ##########################################################################
1>       0.000466: Round: 1
1>       0.000484: Emitting...
1>      20.361530: Done emitting
1>      20.469170: Done
1>     267.956 [  23.448]: Phase: Done                                               [types:   365,  fields:    656,  methods:   1540]
1>     303.124 [  35.168]: Done
1>     303.128 [   0.003]: Saving Results for ARM\Debug\STM32F411\Llilum.Managed
1>     303.130 [   0.002]:     Image...
1>     304.337 [   1.207]:     Image done
1>  Writing LLVM Bitcode representation
1>  Optimizing LLVM Bitcode representation
1>  ===== Alias Analysis Evaluator Report =====
1>    671680 Total Alias Queries Performed
1>    233063 no alias responses (34.6%)
1>    430970 may alias responses (64.1%)
1>    3539 partial alias responses (0.5%)
1>    4108 must alias responses (0.6%)
1>    Alias Analysis Evaluator Pointer Alias Summary: 34%/64%/0%/0%
1>    276253 Total ModRef Queries Performed
1>    118128 no mod/ref responses (42.7%)
1>    420 mod responses (0.1%)
1>    4048 ref responses (1.4%)
1>    153657 mod & ref responses (55.6%)
1>    Alias Analysis Evaluator Mod/Ref Summary: 42%/0%/1%/55%
1>  Compiling LLVM Bitcode
1>     319.707 [  15.370]: Done
1>arm-none-eabi-gcc.exe : error : ARM\Debug\Llilum.Managed_opt.o: No such file or directory`

Well, this time seems a directory miss match thing.
Maybe some compiling parameter inconsistent.
I can find Llilum.Managed_opt.o file in Native\ARM\Debug\STM32F411.

lt72, 2016-09-01:
Will take a look this coming weekend and get back to you by Tuesday morning.

nbsn2, 2016-09-01:
VS + LLILUM SDK

After failure, I copied Llilum.Managed_opt.o file in Native\ARM\Debug\STM32F411 to Native\ARM\Debug,
and run build second time , success.

lt72, 2016-09-01:
interesting, I'll take a look. I bet I forgot to update some strings last time I made changes to the makefile build...

vgolovanov, 2017-01-06:
Seems that bug with Llilum.Managed_opt.o file location is still persists. Can anyone suggest how to fix it ?

Thank you in advance.

FirisSA, 2017-02-09:
I have not find where the "makefile" is generated so I use an other workaround.
Open the frontEnd projet, go in Bench.cs and comment line 1651
`m_outputDir = Path.Combine( m_outputDir,  ExtractProductName( m_compilationSetup.Product ) );`