NETMF/llilum #233: Project compiling issues.

vgolovanov, 2016-09-06:
Hi.
i having problems with compiling LlilumSDK and Llilum project VSIX package.

1st issue, during compilation time, group of referenced libraries (NetmfCompatibilityComponentGroup) from ReferenceAssemblies file in  LlilumSDK not found. But commenting out or removing that section let compile the project.

2nd issue, when building VSIX package, getting error "The "FindInstalledExtension" task was not given a value for the required parameter "InstallableExtensionVersion".	Microsoft.MIEngine.CoreRegisters.ARM"

Any help would be much appreciated.

Best regards,
Vyacheslav.

nbsn2, 2016-09-07:
Just ignore those message.
Go find the .msi and  .vsix file, it's there.

vgolovanov, 2016-09-07:
May be there is some switch to ignore that errors in Visual Studio.
But vsix file is not builded. Msi is builded, after i removed that section.

vgolovanov, 2016-09-08:
Thank you for suggestion. Was looking VSIX file in wrong location.

closed 2016-09-08 by vgolovanov