NETMF/llilum #236: Issue same as #181 #225

vgolovanov, 2017-01-09:
Seems that bug with Llilum.Managed_opt.o file at location "Native\ARM\Debug\STM32F411" is still persists. 

Can anyone suggest how to fix it ?

Thank you in advance.

snikeguo, 2017-01-09:
The llilum project is not update....  ~_~

vgolovanov, 2017-01-09:
I think at some point that project doing what you expecting, For sure it's still full of small bugs, but I hope that somebody know how to fix that bug and will share a fix.

lt72, 2017-01-10:
Hi, the path above comes from the project targets:

```
Cmd>findstr /i /s /p /c:"_opt" *
LlilumApplicationType\Llilum\1.0\Android.Common.targets:  <Target Name="ConvertManagedCode" DependsOnTargets="ResolveReferences" Condition="'@(ReferencePath)' != ''" Inputs="@(ReferencePath)" Outputs="$(IntDir)%(ReferencePath.Filename)_opt.o">
LlilumApplicationType\Llilum\1.0\Android.Common.targets:      <Object Condition="'@(ReferencePath)' != ''" Include="$(IntDir)%(ReferencePath.Filename)_opt.o" />
```

that said, I do not understand exactly how the issue manifest and how to fix it